TGit-Tech/GRBL-28byj-48 #2: Laser Mode

shloma-goldsmidt, 2020-05-17:
Is the Laser Mode operational in this grbl version?

TGit-Tech, 2020-05-24:
I'm not sure;  It's an exact copy of grbl with minor tweaks to the motor signals.  I believe the version can be found in the source code (don't know it off hand).

shloma-goldsmidt, 2020-05-25:
The thing is that when I send $$ command I get the parameter $32 (responsible for laser mode) in the list, so the version of gbrl is 1.1 (as I know laser mode implemented in this version first).

But when I try to enable laser mode (i.e. send $32=1) I encounter with an error message "Homing cycle failure. Homing is not enabled via settings", and the parameter $32 remains the same.

shloma-goldsmidt, 2020-05-25:
The problem with $32=1 has been solved by uncommenting the "VARIABLE_SPINDLE" parameter in config.h. Looks like now the firmware goes into laser mode.

But a new problem appeared: the laser does not turn on (the laser itself is alive because it works directly from a power supply).

My idea is that something wrong with pwm. I do not have an oscilloscope but when I connect a LED between D11 and GND, it does not light up when LaserGRBL runs with S255 command. But when I flash Arduino board with a simplest pwm example, demonstrating analogWrite(), the LED lights up and down smoothly...

TGit-Tech, 2020-10-30:
I've added a folder that tevosyants made changes to in order to activate by pin 12 but it uses grbl 0.8 as I understand it.  Also the digital pins 10 and 11 are used for the motor instead of skipped over.  I haven't had time to test it or know the details of it; but if you'd like to give it try that'd be great.