TGit-Tech/GRBL-28byj-48 #5: [Help] 9g Servo not working.

Crazycarpet-dot, 2020-11-24:
Hey, first off let me just say thanks for releasing a modern verison of grbl that works with unipolar servos.
I'm building an ink plotter and everything mostly works except for one thing.. I'm using a 9g servo (S51 in this case) for testing of raising and lowering the ink dispenser however the servo commands seem to have no effect. 

I tried some modifications myself and couldn't make ti work and I was wondering if you had any suggestions on how I can make this happen? Thanks.

TGit-Tech, 2020-11-27:
Have you tried the firmware at https://github.com/ruizivo/GRBL-28byj-48-Servo  ?  It was setup for Z-Servo if I remember correctly.  This firmware runs a stepper signal on all x,y and z axis.

Crazycarpet-dot, 2020-11-27:
I apologise i forgot i posted this i modified 1.1h based on what you just linked for x y unipolar stepper and z servo. Thanks for the quick reply ill post my version to my git incase anyone needs it im the future.

closed 2020-11-27 by Crazycarpet-dot