TGit-Tech/GRBL-28byj-48 #9: Y and Z vibrating

DJVibezzz, 2021-02-15:
Hello,

When I use this, my X turns both ways but my Y and Z turn only one way and vibrates the other way.
How do I fix this?

TGit-Tech, 2021-02-16:
Sounds like a hardware issue.  If you're using the ULN2003 driver board I cannot think of a way the code could 'vibrate' the stepper in one direction but move in the other since its 4-wire sequential.  However; if you're playing with the code (this code) that would be a different question altogether.  You might want to test your hardware with just some general move code?

DJVibezzz, 2021-02-16:
Thanks for you reply!
I tested it with an arduino uno and it worked fine!
However when I run gcode on it and it moves all 3 axis, then it cuts power while moving.
I am using a 9V 1A power supply

TGit-Tech, 2021-02-22:
You didn't mention which motor driver board you're using.  If you're trying to wire directly to an UNO power loss would be expected as I'm pretty sure motors will overloading the IO pins.

DJVibezzz, 2021-02-22:
I use 3x ULN2003 and it cuts power

TGit-Tech, 2021-06-13:
Sorry its taken so long to get back;  summer work.  Did you find a solution?

DJVibezzz, 2021-06-14:
Hello, I still didn't find my solution

TGit-Tech, 2021-06-14:
What exactly do you mean by 'it cuts power'?  
Have you tried using a different power source like a 12Vdc one?  
And checked the 5-12V pin jumpers on the ULN board for the correct jump?

Does this symptom happen both with gcode execution and manual control?
Is the motor gear / framing snagging or pinching in one direction?  I've put machine oil or cooking oil (pam) on some hardware to get smoother slide and gear mesh..

Those are about the only things I can think to check.

DJVibezzz, 2021-06-14:
It cuts power when all 3 motors are used. The leds also turn off for a second. I didn't try a 12v source because I don't have one. I will try that.

TheExpertNoob, 2022-07-07:
Line 244 of stepper.cpp looks sus.