TGit-Tech/GRBL-28byj-48 #10: Servo motor SG 90

neves1989, 2021-03-06:
I have a small cnc plotter running with 3 5v motors and uln2003 drivers it possible to replace the z axis motor with an SG90 servo?

TGit-Tech, 2021-03-14:
This work was derived from: See https://github.com/ruizivo/GRBL-28byj-48-Servo

#GRBL 28byj-48 + Servo Motor

Should work

Gesiima, 2021-03-14:
How it is possible to use z axis with motor and a servo for other stuff same time?

TGit-Tech, 2021-03-20:
I'm afraid I'll need a little more explanation.

Gesiima, 2021-03-20:
I have printed these Robotarm
[https://www.thingiverse.com/thing:2838859]
and try to use your Firmware.
To open and close the gripper, I need to control the servo.

TGit-Tech, 2026-03-25:
This code isn't for One Servo and two 28BYJ-48 Steppers.
https://github.com/ruizivo/GRBL-28byj-48-Servo is where you'll find a single servo + 2 stepper setup.
This code is for all 3-steppers.

gweep, 2026-03-25:
Thank you,  Yes I knew that.   I was hoping to find out how Geslima solved the problem.