TGit-Tech/GRBL-28byj-48 #15: Step Z VIBRATING

jayanewbie, 2021-08-15:
Please help me 
Step Z VIBRATING
Step Y Normal
Step X Normal
I'am Using Arduino Uno give me schematic for arduino uno Step z part🙏🙏🙏

Carlos-Henreis, 2021-09-07:
I have this problem too, you managed to solve?

Carlos-Henreis, 2021-10-03:
In my case it was a problem with the hardward pin d13 that was poorly soldered and with bad contact,

oluwaseyi-25, 2024-02-16:
I have this same issue. Z is turning clockwise but not anticlockwise

TGit-Tech, 2024-02-21:
I haven't got time to look into this right now but be aware that D13 has an on-board Arduino LED attached to it.
My first instinct would be to add a 2.2K-ohm from D13 to 5V to ensure proper pull-up.

oluwaseyi-25, 2024-02-21:
My problem was resolved already. It happened because some connections were not firm