TGit-Tech/GRBL-28byj-48 #16: credit to costycnc first release

costycnc, 2021-08-24:
hy ... im costycnc , first that i modified grbl from bipolar to unipolar! you give me happy if insert a credit to costycnc.. because your code contain many modifications and variable that i created first .... you can find on github.com/costycnc .... also i modified grbl1.1h see here https://github.com/costycnc/costycnc-costycnc-grbl-1.1-unipolar
for demonstrate that is my first code you can see costyx,costyy variable .... and costyz create by you perhaps without know from where come costy variable .... im happy that my little work is updated by big programmers like you .... but i thing also that is correct that nominated me ! Finally is your decision and i respect your decission is good or not! ... Regards!

TGit-Tech, 2021-09-27:
Your credit will have to come from ruizivo as that's where my initial reference code came from though I do find it odd your branch is all posted within a year or so, mine's within two to three years and ruizivo is clear back from 7 to 11 years ago.  Not that you didn't make those changes for ruizivo code;  just saying your credit will have to come from the person who actually posted it's usage first.

costycnc, 2021-09-30:
You make a big confusion and undocumented ! Anyway with credit or not ... im happy that my changes in first grbl 0.8c is util ... and recent in 1.1h! Anyway ... if variabile costy remain and not delete ... is all respect that i want! Regards!