TGit-Tech/GRBL-28byj-48 #21: Wrong Movements

torstrick, 2024-09-10:
Everything is set up to the point of using UGS to control the plotter. All the jog controls in UGS work correctly, and move the correct direction and distance. Running gcode has unexpected results; the Z axis works correctly. The X and Y axes do not. Trying to draw a simple circle results in seemingly random movements that draws lines in random directions.

torstrick, 2024-09-11:
Resolved this problem by lowering the max rate values. Now motors moved as expected.

Ynauuiu6, 2025-04-15:
Hi, could you share your settings? I also meet the same problem

torstrick, 2025-07-24:
> Hi, could you share your settings? I also meet the same problem

Sorry that I only just noticed this. It's been a while, so I don't remember the exact settings, but I kept various settings files. The one with the lowest values are:

$110 = 200.000	(X-axis maximum rate, mm/min)
$111 = 200.000	(Y-axis maximum rate, mm/min)
$112 = 250.000	(Z-axis maximum rate, mm/min)

If that still doesn't work, keep lowering the values (100, 50, etc. until you find a speed that works). This tells the motors how fast to move, and the default settings were way too high for 28BYJ-48 motors, causing the unusual behavior. 

To display your current settings, type $$ into the terminal when your printer is connected and it will show them. To change a setting, type the $ number with the new value into the terminal. For example, to change the X axis value to 200.000, type "$110 = 200.000" into the console and hit enter. Type $$ to recheck your settings and make sure they were updated.